zang09/ORB_SLAM3_ROS2 #1: CMake Error at ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:100 (message):

wolfgangbiang, 2022-09-06:
Hi,  i think what you are doing is interesting and you did great job. But i facing some problem with compile
 when i run $ colcon build --symlink-install --packages-select orbslam3
I got 
li@li:~/colcon_ws$ colcon build --symlink-install --packages-select orbslam3
Starting >>> orbslam3
--- stderr: orbslam3                                
CMake Error at ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:100 (message):
  ament_cmake_symlink_install_directory() can't find
  '/home/li/colcon_ws/src/orbslam3_ros2/launch'
Call Stack (most recent call first):
  ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:317 (ament_cmake_symlink_install_directory)
  cmake_install.cmake:41 (include)

make: *** [install] Error 1
---
Failed   <<< orbslam3 [1.39s, exited with code 2]
Do you how to deal with it ?

zang09, 2022-09-07:
Hi, wolfgangbiang.
I find error in my `CMakeLists.txt`, so i fix it and commit now.
Please recompile with new code, i hope this will solve your problem.
Thank you.

wolfgangbiang, 2022-09-07:
Hi, zang09
The compiling succeeded,  Thanks for your help.
But i'm facing another problem, when i trying to use ros2 mono 
The commend is 
ros2 run orbslam3 mono /home/li/catkin_ws/src/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/li/catkin_ws/src/ORB_SLAM3/Examples/Monocular/RealSense_D435i.yaml
the result is 

[image]
 It seems failed to load ORBvoc.txt
Do you have any ideas about that ?

zang09, 2022-09-07:
I think that was problem of yaml config file. (but not sure)
Did you try with the ros version of config file? (Like this : [link](https://github.com/zang09/ORB-SLAM3-STEREO-FIXED/blob/master/Examples_old/Monocular/RealSense_D435i.yaml))
Or you can try with other yaml config file.

closed 2022-09-20 by zang09