zang09/ORB_SLAM3_ROS2 #2: ORB-Slam3 for ROS2 stops after receiving 1 frame

sumitpaulde, 2022-09-08:
Thanks a lot for this beautiful work. I am publishing the sensor.image/Image type message from my one ROS2 node and trying to run ORB-Slam3 for the published topics. I have also edited the topic name in the monocular-slam-node.cpp. Even though I can see the stream of images as a video(continous frames) from the subscriber, the ORBSLAM3-for-ROS2 stops working as soon as it receives one frame. Any idea what I am doing wrong?
[image]
[image]

zang09, 2022-09-13:
I guess it was problem of less feature extraction from your camera image.
I may recommended test it your build program first, on given dataset.

sumitpaulde, 2022-09-13:
This camera image works fine with the ROS1 ORB slam but not with the ORB slam for ros2. Do you have any idea what is wrong?

zang09, 2022-09-13:
Then, i don't know what is the problem.. exactly. I need to test it on the same evironment to figure out.

ZhaoDidididi, 2022-09-16:
I have the same problem.

ZhaoDidididi, 2022-09-20:
> This camera image works fine with the ROS1 ORB slam but not with the ORB slam for ros2. Do you have any idea what is wrong?

I solved the problem by replacing the default version 4.5.5 OpenCV of Jetpack 5.0.2 with the version 4.2.0 OpenCV. For me, the problem is the version conflict of the cv-bridge. The default version of the cv-bridge in ROS2 box is version 4.2.0, so the ORB-SLAM3 source code needs to be compiled using OpenCV4.2.0.

zang09, 2022-09-20:
Hi, @ZhaoDidididi.
It's good to here that you solved the problem.
I didn't trouble with this problem, because i tested it on `OpenCV 4.2.0`.
I will write this troubleshooting on `Readme.md`. Thank you.

XCYYYYY, 2022-10-13:
> I solved the problem by replacing the default version 4.5.5 OpenCV of Jetpack 5.0.2 with the version 4.2.0 OpenCV. For me, the problem is the version conflict of the cv-bridge. The default version [...]

hello,i solved this problem by replacing opencv4.5.0 with opencv 4.2.0 when i run mono mode, but when i run the RGBD mode i still face the problem: "ORBSLAM3-for-ROS2 stops working as soon as it receives one frame", i am confused, do u have any idea?
thanks a lot!

EndlessLoops, 2023-03-28:
https://www.ncnynl.com/archives/202107/4491.html

closed 2023-04-03 by zang09