zang09/ORB_SLAM3_ROS2 #4: stereo-inertial node can't run well

wolfgangbiang, 2023-02-23:
Hey zang, have you successfully run stereo-inertial mode in realtime?
By my side it also output IMU not initialzation and Empty IMU Measurement Vector!!
Do you have any idea how to solve this?
[image]

lfescobarc, 2023-02-23:
I am having the same issue, I am trying to run using a D435i realsense camera.

zang09, 2023-02-23:
Hi, i have successfully run in real-time with simulator dataset.
Did you compile with this [repo](https://github.com/zang09/ORB-SLAM3-STEREO-FIXED)? The fixed version of ORB_SLAM3 stereo mode.

wolfgangbiang, 2023-02-23:
No i do it on original ORB-SLAM3, but by my side , the stereo model works completely fine, The issue happen only when Stereo-Inertial model. Anyway i will try this repo first. Thanks.

wolfgangbiang, 2023-02-23:
I also use the D455I and even i changed to the orb-slam3-fixed, it still the same condition.  The tracking thread can't work, IMU can't be initialized. I checked ros2 topic of imu. Data streams are there and subscriber topic is correct. Do you have any idea about that?

lfescobarc, 2023-02-23:
I am working with stereo-inertial as well. I tried with both, the repo you shared with the fixed ORB-SLAM3 and the official one, I also tested a D455 and a D435i camera, checked the ROS2 rqt_graph and used the infra1 and infra2 as left and right camera, it looks like camera/imu is also connected to ORB-SLAM, howevere it still not shows any image, from what I tested the ORB-SLAM examples for stereo-inertial, and they seem to work. Any idea?

wolfgangbiang, 2023-02-24:
> 

Did you check your D435i.yaml file? Calibration parameters may cause your problem.

lfescobarc, 2023-02-24:
> > 
> 
> Did you check your D435i.yaml file? Calibration parameters may cause your problem.

Yes, I did, what I am not sure is if I need to rectify or not the image when using the Infra1 and 2 as left and right. I am using the yaml file of the 435 as it is in the config folder, and from what I get, ORB SLAM3 should be showing something even with the parameters a little of.

lfescobarc, 2023-02-28:
> 

I can try, Do you mean an actual router or something on the code?

wolfgangbiang, 2023-03-01:
> I can try. Do you mean an actual router or something on the code? sorry for the basic question, but I do not understand what you mean with router there.

WLAN router

closed 2023-04-03 by zang09

LIBOZH97, 2023-07-18:
I met the same problem,i tried to combine ros2 orbslam3 with gazebo,when i run the code,there was no output on the orbslam3,but the ralation of topic is correct,i really don't know what going on.