zang09/ORB_SLAM3_ROS2 #5: /usr/bin/ld: /usr/local/lib/libopencv_calib3d.so.407: error adding symbols: DSO missing from command line

weiwei-dotcom, 2023-04-03:
[image]
what's wrong with this error?

1347849769, 2023-04-07:
I have the same question

1347849769, 2023-04-09:
> [image]
> what's wrong with this error?
the ros2 version must be foxy,

> [image]
> what's wrong with this error?
This is not a problem with opencv. Maybe your ros2 version is not foxy. I changed the Ubuntu system to 20.04 and used the foxy version of ros2 to compile normally

1347849769, 2023-04-09:
> [image]
> 这个错误有什么问题？
This is not a problem with opencv. Maybe your ros2 version is not foxy. I changed the Ubuntu system to 20.04 and used the foxy version of ros2 to compile normally

EndlessLoops, 2023-04-17:
https://github.com/alsora/ros2-ORB_SLAM2/issues/8#issuecomment-1461570970

closed 2023-09-27 by zang09

hermanumrao, 2025-05-25:
> I fixed it and created a pull request, try it if you are still interested

I was gtting the same error
@Fran-FC  didn't see you pull request,
but then i just went your profile and copied your [cmakelists.txt](https://github.com/Fran-FC/ORB_SLAM3_ROS2/blob/main/CMakeLists.txt)  and now it works fine

thanks man