zang09/ORB_SLAM3_ROS2 #11: segmentation fault while running for mono/stereo camera using realsense d455i

raviS123-dot, 2023-06-28:
v3@v3:~/ORB_SLAM3$ ros2 run orbslam3 mono /home/v3/colcon_ws/src/orbslam3_ros2/vocabulary/ORBvoc.txt /home/v3/colcon_ws/src/orbslam3_ros2/config/rgb-d/TUM1.yaml 

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- Image scale: 1
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- p1: -0.005358
- p2: 0.002628
- k3: 1.16331
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
[ros2run]: Segmentation fault

Fran-FC, 2023-06-28:
I am facing the same error, which version of ros are you using?

raviS123-dot, 2023-06-28:
@Fran-FC ros2 humble

astronaut71, 2023-07-06:
same problem here. Solve the problem? Im running ros2 humble

Nikolaiarn, 2023-11-11:
Also have the same problem with ros2 Humble. Has anyone solved it?

Nikolaiarn, 2023-11-14:
@astronaut71 @raviS123-dot @Fran-FC Has anyone found a solution to this problem?