zang09/ORB_SLAM3_ROS2 #13: having problem on real time

LIBOZH97, 2023-07-18:
[image]
i want to combine gazebo and ros2_orbslam3.
on the photo,the image should be represented on orb slam3,but it doesn't work,i don't know how to solve this problem.

LIBOZH97, 2023-07-18:
and the topic is correct,...

Ailce88899, 2023-07-19:
Refer to this article:
https://www.ncnynl.com/archives/202107/4492.html

Narek-939, 2023-09-16:
Hi everyone, I have the same problem, my screen was black when I subscribed images, https://www.ncnynl.com/archives/202107/4492.html 
I tried to understand , but I couldnt, please explain me what I need to do, or if you have solved this, please point me the way)) thanks)

LIBOZH97, 2023-09-16:
i solved the problem，try to move cam in a complicated environment（with lots of lines or points）so orbslam3 could get enough feature points。and do not stop moving cam 

淦
***@***.***

Narek-939, 2023-09-16:
I tried but didnt have any result, I tried to change the number of keypoints in yaml file, but it doesnt work yet

LIBOZH97, 2023-09-16:
截个图看看 

淦
***@***.***

Narek-939, 2023-09-16:
[image]

LIBOZH97, 2023-09-16:
用ros2话题命令查看一下订阅到了没有，怎么感觉像是没有订阅到话题呢，订阅到了之后命令行应该会显示 frame has been sent，检查下 

淦
***@***.***

Narek-939, 2023-09-16:
[image]

Narek-939, 2023-09-16:
[image]

LIBOZH97, 2023-09-16:
你订阅的是电脑自带的摄像头吗，如果是，移动一下电脑，在那种边边角角比较多的，复杂的地方，多移动一段时间
 

淦
***@***.***

Narek-939, 2023-09-16:
I tried but didnt have any result

Narek-939, 2023-09-16:
I have showed my image before giving to

m_SLAM->TrackMonocular(image, Utility::StampToSec(msg->header.stamp));

and it was normal image, so I think that it was from ORB_SLAM3 system