zang09/ORB_SLAM3_ROS2 #15: colcon build failed undefined reference to `DUtils::Random::SeedRandOnce()'

spectra543, 2023-07-28:
Hi guy i try to colcon build ORB_SLAM3_ROS2 after i run this line  "colcon build --symlink-install --packages-select orbslam3" this error happens.

[image]

i use 
ros2 foxy
opencv 4.2.0
ubuntu 20.04
anyone know how to fix thankyou!

mjunsen123, 2024-04-26:
According to the solution from [https://blog.csdn.net/weixin_44401286/article/details/111468851](url)

1. Go to ORB_SLAM3/Thirdparty/DBoW2/CMakeLists.txt
2. Change line `add_library(DBoW2 DYNAMIC ${SRCS_DBOW2} ${SRCS_DUTILS})` to `add_library(DBoW2 STATIC ${SRCS_DBOW2} ${SRCS_DUTILS})` and,
3. add line `set_property(TARGET DBoW2 PROPERTY POSITION_INDEPENDENT_CODE ON)`
4. Go to ORB_SLAM3/CMakeLists.txt
5. Replace `${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.so` with `${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.a`
6. Rebuild

Hope this help!

1melek, 2024-05-14:
i have the same issue , i tried  doing your method and it still didn't work

mjunsen123, 2024-05-14:
> i have the same issue , i tried doing your method and it still didn't work

Have you rebuild ORBSLAM3 as well? Do you have `.a` file in Thirdparty/DBoW2/lib/libDBoW2.a after rebuild?

1melek, 2024-05-14:
well no i will try to rebuild it now

1melek, 2024-05-14:
okay that worked ! but now the build failed again.. 

[image]

 can you help?

mjunsen123, 2024-05-14:
> okay that worked ! but now the build failed again..
> 
> ![Screenshot from 2024-05-14 12-20-23](https://private-user-images.githubusercontent.com/144792577/330565797-ca193a90-25f6-4ffa-b98a-18b835716579.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU3MTQ4NjEsIm5iZiI6MTcxNTcxNDU2MSwicGF0aCI6Ii8xNDQ3OTI1NzcvMzMwNTY1Nzk3LWNhMTkzYTkwLTI1ZjYtNGZmYS1iOThhLTE4YjgzNTcxNjU3OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxNFQxOTIyNDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zM2QyNzczMzg2NTUyMThlYzYyOGU3ZDUyODlkN2M5NjU5MzRjN2UzYjk4OTI3MzUyYWJjYmE2ZTkwYzhkMmVhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.yzFcEXiNu_6rbtbsOzfYEJ57AshhQRyOBWkqXDjWQa8)
> 
> can you help?

Your OpenCV version in ROS is different with the one in your machine. I assume you have build cvbridge, edit the file CMakeList.txt in CVBridge to force it to use OpenCV 4.9.

1melek, 2024-05-14:
no i haven't build CVBbridge ..

mjunsen123, 2024-05-14:
Okay if it is the case i suggest you downgrade your opencv in your ubuntu to 4.2

1melek, 2024-05-14:
the thing is , i have opencv 4.2 and i have no idea where 4.9 is coming from

mjunsen123, 2024-05-20:
Have you build with source before?

1melek, 2024-05-20:
I figured that and it build successfully but now when i run it i can't see anything in the camera and I don’t know from where it get camera input , like it always says “1 camera in the atlas” even when i deactivated the laptop camera and no usb cam

andremaurell, 2024-07-02:
@1melek Good tutorial my friend, but, can you help-me? after the changes at CMake and generate .a file, I received that error:

[image]

mjunsen123, 2024-07-02:
Never see this error before. Can you please confirm you have CVBridge (vision_opencv) installed?

andremaurell, 2024-07-02:
I think yes:

[image]

mjunsen123, 2024-07-02:
And can you confirm that you have same version of OpenCV in both your ROS and Ubuntu?

andremaurell, 2024-07-02:
i don't know how I can see on ROS, but on ubuntu is 4.5.4

mjunsen123, 2024-07-03:
I am not sure but I think it may due to different version of openCV in Ros and Ubuntu. I am aware of this but I am not sure if this is your case. What I did is I build opencv and cvbridge from source, so I can specific the version used in ORBSLAM3

andremaurell, 2024-07-03:
Yeah, I think is the same version, but I'm not sure about that. Do you have same tutorial showing how I can make this build?