zang09/ORB_SLAM3_ROS2 #17: question regarding the installation of  humble branch and opencv 4.5.4

tschinfra, 2023-09-13:
Hi @zang09,

i have a question regarding the installation.
You mentioned in the Prerequisites, that you tested the humble-branch with:

>Ubuntu 22.04
>ROS2 Humble
>OpenCV 4.5.4
>Build ORB_SLAM3

How did you installed than ORB_SLAM3 ? Which OpenCV version did you used? 4.0 or your tested 4.5.4?

Tanks in advance!

kimh060612, 2023-12-28:
you can clone the ORB_SLAM3 in c++14_comp branch
It is compatible with C++14 and OpenCV 4.5.4
Before you run, make sure that you must reinstall all the dependencies in order that you rebuild them from opencv 4.5.4!
If not, it may cause the segmentation fault error

andremaurell, 2024-07-04:
hey @kimh060612 can you help-me in the step of reinstall the dependencies from opencv 4.5.4?

kimh060612, 2024-07-13:
For convenience, I wrote Dockerfile for ORB_SLAM's dependencies.
Plz check my repository: https://github.com/DR4R/ORB_SLAM3_Wrapper.git
Even if you want to install on your host, it may help you