zang09/ORB_SLAM3_ROS2 #20: Segmentation fault when running ros2 humble (mono)

Nikolaiarn, 2023-11-14:
I have seen that there is someone else who has had the same error, but I cant find any solution. OrbSlam3 works fine in the euroc demo. Here is the full error:

nikolai@NikUbuntu22:~/colcon_ws$ ros2 run orbslam3 mono /home/nikolai/colcon_ws/src/orbslam3/vocabulary/ORBvoc.txt /home/nikolai/colcon_ws/src/orbslam3/config/monocular/TUM1.yaml 

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- Image scale: 1
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- p1: -0.005358
- p2: 0.002628
- k3: 1.16331
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
There are 1 cameras in the atlas
Camera 0 is pinhole
slam changed
============================ 
Starting the Viewer
[ros2run]: Segmentation fault

Nikolaiarn, 2023-11-17:
You can solve the problem by unninnsistalling all ORB-SLAM files, then reverting back to opencv 4.2.0 then reinstalling all the files. worked for me

Sn0wl3r0ker, 2023-12-02:
> You can solve the problem by unninnsistalling all ORB-SLAM files, then reverting back to opencv 4.2.0 then reinstalling all the files. worked for me

worked for me, thank you!

TSoli, 2024-03-01:
> You can solve the problem by unninnsistalling all ORB-SLAM files, then reverting back to opencv 4.2.0 then reinstalling all the files. worked for me

Amazing! Worked for me too! I had to disable tbb when compiling opencv 4.2.0 with `-DWITH_TBB=OFF` on Linux Mint 21.3 (based on Ubuntu 22.04).

dyhuachi, 2024-07-23:
> You can solve the problem by unninnsistalling all ORB-SLAM files, then reverting back to opencv 4.2.0 then reinstalling all the files. worked for me

why the opencv 4.8.1 can not run,this question can be fixed?

LRaina215, 2025-07-30:
Worked for me!!! Thankkk u soo much!!!!