zang09/ORB_SLAM3_ROS2 #21: Does anyone have folder for monocular inertial (ROS2 Humble)

Nikolaiarn, 2024-01-22:
I see that there are folders that have scripts for inertial, rgbd, sterio and sterio inertial. I believe that it should be possible for obrslam3 to handle monocular inertial data. I was wondring if any of you have the three C++ scripts for this implementation?

astronaut71, 2024-03-06:
I tried but not working.

Haneenalnahwi, 2024-05-02:
I need this too. Is there any update on mono-inertial?

Nikolaiarn, 2024-05-03:
@Haneenalnahwi, I found that the fork:  https://github.com/kyrikakis/ORB_SLAM3_ROS2.git Had a working version for mono VI slam. Personally I had some problems with it since I was not able to extract the IMU parameters from kalibr since it is using ros1. I wish you good luck

Nikolaiarn, 2024-05-06:
@Haneenalnahwi, somone just posted a repo with working VI-SLAM in ROS2 if you are still interested:
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/793#issuecomment-2095639783