zang09/ORB_SLAM3_ROS2 #24: Error running rgb-d SLAM ROS2 (humble): double free or corruption (out) [ros2run]: Aborted

RodrigofrBastos, 2024-04-02:
### System Setup:
* ROS2 Humble
* Ubuntu 22.04
* Simulated RGB-D camera (gazebo Ignition)

When setting everything to run ORBSLAM3, trying to run `$ ros2 run orbslam3 rgbd PATH_TO_VOCABULARY PATH_TO_YAML_CONFIG_FILE` and receiving the following output:
```
Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
Shutdown

Saving keyframe trajectory to KeyFrameTrajectory.txt ...
double free or corruption (out)
[ros2run]: Aborted
```
For solving this error what is needed to do?

RodrigofrBastos, 2024-04-02:
After searching for a while, I found a website that had that answer, basically you need to replace https://github.com/zang09/ORB_SLAM3_ROS2/blob/00c54335ccc010d74c1e24e336aa817604124947/src/rgbd/rgbd-slam-node.cpp#L11C1-L12C122
Replace with the following lines:
```
rgb_sub = std::make_shared<message_filters::Subscriber<ImageMsg> >(this, "/camera/rgb");
depth_sub = std::make_shared<message_filters::Subscriber<ImageMsg> >(this, "/camera/depth");
```

RodrigofrBastos, 2024-04-02:
Someone had the same problem?

TSoli, 2024-04-22:
Yes this fixed it for me, thanks very much!

fishyy119, 2025-01-08:
This solution also solved my problem, and I'm curious as to why this is happening.

liuze1111, 2025-04-13:
感谢！