zang09/ORB_SLAM3_ROS2 #26: double free or corruption (out) [ros2run]: Aborted

li-dahua, 2024-05-30:
**ROS2 Humble
Ubuntu 22.04**
I tried this already. Not working.
```

rgb_sub = std::make_shared<message_filters::Subscriber<ImageMsg> >(this, "/camera/rgb");
depth_sub = std::make_shared<message_filters::Subscriber<ImageMsg> >(this, "/camera/depth");

```
The program crashes without showing output.
ros2 run orbslam3 stereo vocabulary/ORBvoc.txt BFS.yaml False

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name:

Camera Parameters:

Camera: Pinhole
Image scale: 0.25
fx: 1749.73
fy: 1764.11
cx: 255.644
cy: 114.834
k1: -0.00412
k2: 0.723251
p1: -0.007554
p2: 0.001609
k3: 0
fps: 30
color order: BGR (ignored if grayscale)
Depth Threshold (Close/Far Points): 50.0339

ORB Extractor Parameters:

Number of Features: 1250
Scale Levels: 8
Scale Factor: 1.2
Initial Fast Threshold: 20
Minimum Fast Threshold: 7
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
Shutdown
Saving keyframe trajectory to KeyFrameTrajectory.txt ...
double free or corruption (out)
[ros2run]: Aborted

IaroslavS, 2024-07-10:
I faced the same problem. I solved it with correcting node class. I created node outside the class in the main (stereo.cpp), then put the created node variable into the node class (StereoSlamNode). It looks like StereoSlamNode class cannot access the shared_ptr<rclcpp::Node>(this) inside the StereoSlamNode constructor, thus instead of shared_ptr<rclcpp::Node>(this) an external node should be used.
Here's my proposed solution:

stereo.cpp
```cpp
    rclcpp::init(argc, argv);
    if(argc < 4)
    {
        std::cerr << "\nUsage: ros2 run orbslam stereo path_to_vocabulary path_to_settings do_rectify" << std::endl;
        return 1;
    }
    auto node = std::make_shared<rclcpp::Node>("run_slam");

    // malloc error using new.. try shared ptr
    // Create SLAM system. It initializes all system threads and gets ready to process frames.

    bool visualization = true;
    ORB_SLAM3::System pSLAM(argv[1], argv[2], ORB_SLAM3::System::STEREO, visualization);

    std::shared_ptr<StereoSlamNode> slam_ros;
    slam_ros = std::make_shared<StereoSlamNode>(&pSLAM, node.get(), argv[2], argv[3]);
    std::cout << "============================ " << std::endl;

    rclcpp::spin(slam_ros->node_->get_node_base_interface());
    rclcpp::shutdown();

    return EXIT_SUCCESS;
```

stereo-slam-node.hpp
```cpp
class StereoSlamNode : public rclcpp::Node
{
public:
    StereoSlamNode(ORB_SLAM3::System* pSLAM, rclcpp::Node* node, const string &strSettingsFile, const string &strDoRectify);

    ~StereoSlamNode();

    rclcpp::Node* node_;
```

stereo-slam-node.cpp
```cpp
StereoSlamNode::StereoSlamNode(ORB_SLAM3::System* pSLAM, rclcpp::Node* node, const string &strSettingsFile, const string &strDoRectify)
:   Node("ORB_SLAM3_ROS2"),
    m_SLAM(pSLAM),
    node_(node)
{
    left_sub = std::make_shared<message_filters::Subscriber<ImageMsg> >(node_, "camera/left");
    right_sub = std::make_shared<message_filters::Subscriber<ImageMsg> >(node_, "camera/right");
    syncApproximate = std::make_shared<message_filters::Synchronizer<approximate_sync_policy> >(approximate_sync_policy(10), *left_sub, *right_sub);
    syncApproximate->registerCallback(&StereoSlamNode::GrabStereo, this);
}
```

IaroslavS, 2024-07-10:
#24 
https://github.com/zang09/ORB_SLAM3_ROS2/issues/24