zang09/ORB_SLAM3_ROS2 #32: Modification of repository so that this ROS2 wrapper can be used in navigation.

jnskkmhr, 2024-07-10:
Hi, people. 
I modified the original codebase so that I could use it inside navigation. 
https://github.com/jnskkmhr/orbslam3/tree/main

Here is what I did: 

* Modified the node to provide topics (tf2, tracked image, pose topics)
* The stereo-inertial node did not work because of QOS setting in IMU subscriber, so I fixed this.
* SLAM pose was an optical frame pose expressed in OpenCV frame, but I needed camera frame pose in ROS FLU coordinate.
* The node provide map to odom (localization) transform

Currently, unfortunately, the stereo-inertial node does not work properly. 
I see IMU initialization takes a lot of time, and the pose sometimes jumps.
This might be because of the wrong calibration configuration.
Please send PR if you know how to solve issues.

NilsRublein, 2024-07-15:
I tried to install your version, but I am stuck at a dependency problem: It seems that `CmakeLists.txt` cannot find the `epoxy` library. 

```
colcon build --cmake-args -DCMAKE_CXX_FLAGS="-w" --symlink-install --packages-select orbslam3
Starting >>> orbslam3
--- stderr: orbslam3                         
CMake Error at CMakeLists.txt:33 (find_package):
  By not providing "Findepoxy.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "epoxy", but
  CMake did not find one.

  Could not find a package configuration file provided by "epoxy" with any of
  the following names:

    epoxyConfig.cmake
    epoxy-config.cmake

  Add the installation prefix of "epoxy" to CMAKE_PREFIX_PATH or set
  "epoxy_DIR" to a directory containing one of the above files.  If "epoxy"
  provides a separate development package or SDK, be sure it has been
  installed.

gmake: *** [Makefile:528: cmake_check_build_system] Error 1
---
Failed   <<< orbslam3 [0.69s, exited with code 2]
                                
Summary: 0 packages finished [1.09s]
  1 package failed: orbslam3
  1 package had stderr output: orbslam3

jnskkmhr, 2024-07-15:
@NilsRublein 
Did you use docker to build the package ? 
Could you also check if [this path](https://github.com/jnskkmhr/orbslam3/blob/24ae84352df8c72d272432576ed2006e8a971c3c/CMakeLists.txt#L11) is same in your case? (Where you cloned [ORB-SLAM3](https://github.com/jnskkmhr/ORB-SLAM3-STEREO-FIXED.git)) 
If you are using docker, it might be other issues though.

hermanndererdmann, 2024-07-17:
not sure where i'm supposed to run colcon build, but neither on the machine nor in the container it works.

jnskkmhr, 2024-07-17:
@hermanndererdmann 
Hey, apologies for the confusion. 
I modified README so that it is clear where to run the colcon build. 

By the way, I recommend that you use docker as it is difficult for me to figure out some dependency issues that some people may face. 
I did a clean install of orbslam3 for sanity check again, and everything works just fine.

hermanndererdmann, 2024-07-18:
This is what i get. (no realsense connected, just trying to launch the package)

ubuntu@hp:~/Downloads/test/ros/src/orbslam3$ ./docker/build_image.sh
docker build /home/ubuntu/Downloads/test/ros/src/orbslam3/docker --tag orbslam3-ros2 -f /home/ubuntu/Downloads/test/ros/src/orbslam3/docker/dockerfile-foxy
[+] Building 1.8s (23/23) FINISHED                               docker:default
 => [internal] load build definition from dockerfile-foxy                  0.0s
 => => transferring dockerfile: 2.32kB                                     0.0s
 => [internal] load metadata for docker.io/osrf/ros:foxy-desktop           1.4s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => [ 1/19] FROM docker.io/osrf/ros:foxy-desktop@sha256:ce6126b2edd2231ed  0.0s
 => CACHED [ 2/19] RUN apt-get update &&     apt-get install -y --no-inst  0.0s
 => CACHED [ 3/19] WORKDIR /home/thirdparty                                0.0s
 => CACHED [ 4/19] RUN git clone https://github.com/stevenlovegrove/Pango  0.0s
 => CACHED [ 5/19] RUN git clone https://github.com/jnskkmhr/ORB-SLAM3-ST  0.0s
 => CACHED [ 6/19] RUN cd ORB_SLAM3 && git pull origin && chmod +x build.  0.0s
 => CACHED [ 7/19] RUN cd ORB_SLAM3/Thirdparty/Sophus/build && make insta  0.0s
 => CACHED [ 8/19] RUN mkdir -p /etc/apt/keyrings && curl -sSf https://li  0.0s
 => CACHED [ 9/19] RUN apt-get update && apt-get install -y apt-transport  0.0s
 => CACHED [10/19] RUN echo "deb [signed-by=/etc/apt/keyrings/librealsens  0.0s
 => CACHED [11/19] RUN apt-get update && apt-get install -y librealsense2  0.0s
 => CACHED [12/19] WORKDIR /home/thirdparty/ros2_ws/src                    0.0s
 => CACHED [13/19] RUN git clone https://github.com/IntelRealSense/realse  0.0s
 => CACHED [14/19] WORKDIR /home/thirdparty/ros2_ws                        0.0s
 => CACHED [15/19] RUN rosdep install -i --from-path src --rosdistro foxy  0.0s
 => CACHED [16/19] RUN /bin/bash -c "source /opt/ros/foxy/setup.sh && col  0.0s
 => CACHED [17/19] WORKDIR /home/ros2_ws                                   0.0s
 => CACHED [18/19] RUN echo "source /opt/ros/foxy/setup.bash" >> /root/.b  0.0s
 => CACHED [19/19] RUN echo "source /home/thirdparty/ros2_ws/install/setu  0.0s
 => exporting to image                                                     0.0s
 => => exporting layers                                                    0.0s
 => => writing image sha256:8b7cfc5238fd6f98958b8ac986b628dea4df460e4f8df  0.0s
 => => naming to docker.io/library/orbslam3-ros2                           0.0s

 1 warning found (use --debug to expand):
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 2)
ubuntu@hp:~/Downloads/test/ros/src/orbslam3$ ./docker/run_container.sh
non-network local connections being added to access control list
docker: Error response from daemon: Conflict. The container name "/orbslam3-ros2-container" is already in use by container "81fae4d3b3135d56929da1b9794d5d98fcae8c52f3c9d1a51a0c5a5092858ee3". You have to remove (or rename) that container to be able to reuse that name.
See 'docker run --help'.
ubuntu@hp:~/Downloads/test/ros/src/orbslam3$ ./docker/run_container.sh
non-network local connections being added to access control list
root@hp:/home/ros2_ws# colcon build --cmake-args -DCMAKE_CXX_FLAGS="-w" --symlink-install --packages-select orbslam3
[0.335s] WARNING:colcon.colcon_core.package_selection:ignoring unknown package 'orbslam3' in --packages-select
                     
Summary: 0 packages finished [0.12s]
root@hp:/home/ros2_ws# ls
build  install  log
root@hp:/home/ros2_ws# source install/setup.bash 
root@hp:/home/ros2_ws# ros2 launch realsense2_camera rs_launch.py enable_infra1:=true enable_infra2:=true enable_accel:=true enable_gyro:=true unite_imu_method:=2 infra_width:=640 infra_height:=480 camera_name:=d455 camera_namespace:=d455
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-07-18-08-56-26-586723-hp-62
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [64]
[realsense2_camera_node-1] [INFO] [1721292986.869427926] [d455.d455]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1721292986.869600847] [d455.d455]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1721292986.869615324] [d455.d455]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [WARN] [1721292986.876328754] [d455.d455]: No RealSense devices were found!
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[realsense2_camera_node-1] [INFO] [1721292989.137370221] [rclcpp]: signal_handler(signal_value=2)
[INFO] [realsense2_camera_node-1]: process has finished cleanly [pid 64]
root@hp:/home/ros2_ws# ros2 launch orbslam3 -
--debug                         --print-description             --show-arguments                -n                              
--noninteractive                --show-all-subprocesses-output  -a                              -p                              
--print                         --show-args                     -d                              -s                              
root@hp:/home/ros2_ws# ros2 launch orbslam3 stereo_inertial_d455.launch.yaml
Package 'orbslam3' not found: "package 'orbslam3' not found, searching: ['/home/thirdparty/ros2_ws/install/realsense2_description', '/home/thirdparty/ros2_ws/install/realsense2_camera', '/home/thirdparty/ros2_ws/install/realsense2_camera_msgs', '/opt/ros/foxy']"
root@hp:/home/ros2_ws#

NilsRublein, 2024-07-19:
> @NilsRublein Did you use docker to build the package ? Could you also check if [this path](https://github.com/jnskkmhr/orbslam3/blob/24ae84352df8c72d272432576ed2006e8a971c3c/CMakeLists.txt#L11) is same in your case? (Path to [ORB-SLAM3](https://github.com/jnskkmhr/ORB-SLAM3-STEREO-FIXED.git)/CMakeModules) If you are using docker, there might be other issues.

@jnskkmhr I just tried to build it myself without the docker. In my case the path was not correct to the ORBSLAM3 repo, As you forked from [this](https://github.com/zang09/ORB_SLAM3_ROS2) repo, I expected you also used [this](https://github.com/zang09/ORB-SLAM3-STEREO-FIXED) version of ORBSLAM3. After fixing the path and some dependencies (`sophus` & `OpenCV`) I managed to build and run it!

jnskkmhr, 2024-07-19:
@NilsRublein 
That's good to hear. 
On my side, I also forked ORB-SLAM3-Stereo-Fixed repo as I failed to build it. 
https://github.com/jnskkmhr/ORB-SLAM3-STEREO-FIXED
I remember it was something related to find_package(epoxy) and find_package(OpenGL).

NamTruongTran, 2024-08-03:
@jnskkmhr Thank you for your great contributions.

Can I also use your repository (https://github.com/jnskkmhr/orbslam3/tree/main) with Humble?

jnskkmhr, 2024-08-06:
@NamTruongTran 
Hi, I have not tested the repository with Humble yet. 
You need to change the version of OpenCV [here](https://github.com/jnskkmhr/ORB-SLAM3-STEREO-FIXED/blob/4a30e081e2d40af8794cbab743dbcd8d4bccf5ed/CMakeLists.txt#L33) (I think apt in Ubuntu22.04 installs 4.5.4) or source build OpenCV4.2 on Ubuntu22.04 with humble.

NamTruongTran, 2024-08-06:
@jnskkmhr 

Thank you so much for you reply i really appreciate your time.
I want to ./build.sh this:
https://github.com/zang09/ORB-SLAM3-STEREO-FIXED

I got the following error message:
`(foxy) jetsonagx@ubuntu:~/workspaces/orb_slam_ws/src/ORB_SLAM3$ ./build.sh
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jetsonagx/workspaces/orb_slam_ws/src/ORB_SLAM3/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jetsonagx/workspaces/orb_slam_ws/src/ORB_SLAM3/Thirdparty/g2o/build
[100%] Built target g2o
Configuring and building Thirdparty/Sophus ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jetsonagx/workspaces/orb_slam_ws/src/ORB_SLAM3/Thirdparty/Sophus/build
[  8%] Built target test_so2
[ 16%] Built target test_common
[ 25%] Built target test_rxso2
[ 33%] Built target test_geometry
[ 41%] Built target test_se2
[ 50%] Built target test_sim2
[ 58%] Built target test_velocities
[ 66%] Built target test_sim3
[ 75%] Built target test_so3
[ 83%] Built target HelloSO3
[ 91%] Built target test_rxso3
[100%] Built target test_se3
Uncompress vocabulary ...
Configuring and building ORB_SLAM3 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++14.
OPENCV VERSION:
4.2.0
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jetsonagx/workspaces/orb_slam_ws/src/ORB_SLAM3/build
[ 51%] Built target g2o
make[2]: *** No rule to make target '/usr/lib/libOpenNI.so', needed by '../lib/libORB_SLAM3.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:96: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
`

"I have been stuck for almost 2 days now."
Do you have any solution for that?

Thank you !

jnskkmhr, 2024-08-10:
Please use this: https://github.com/jnskkmhr/ORB-SLAM3-STEREO-FIXED instead of https://github.com/zang09/ORB-SLAM3-STEREO-FIXED
I modified couple lines in source code and CMakeLists.txt as there were dependency errors. 

Also please refer to this: https://github.com/jnskkmhr/orbslam3/pull/1
I do not want to deal with host install issues. 
That is why I highly recommend that you use docker. 
In your case, you need to base your docker image on jetson ubuntu 20.04.

adamanov, 2024-08-14:
@jnskkmhr 

just tested on Ubuntu 22.04 with ros2 humble, the nodes are started and working, however node conduct a deep test on if the published odometry correctly, (will back soon)

however, inside the bellow function, you referencing ` m_SLAM->GetCurrentFrame()` which actual do not exist. I believe you make a change inside https://github.com/jnskkmhr/ORB-SLAM3-STEREO-FIXED ? if so, could you push the latest one? 

[publish_tracking_img(pubTrackImage_, this->get_clock()->now(), m_SLAM->GetCurrentFrame(), world_frame);](https://github.com/jnskkmhr/orbslam3/blob/main/src/stereo/stereo-slam-node.cpp#match-159-4)

jnskkmhr, 2024-08-16:
@adamanov 
Hi, thanks for the contribution. 
I think the current odometry has some scale issues. 
I still have not figured out this problem yet, so I highly welcome your PR in the future.

irfanalimd, 2024-11-13:
@jnskkmhr Does the stereo-inertial mode work in your modified repo now ? (I am looking to run it on Ubuntu 22.04 and ROS 2 humble). I have tried running it using @zang09's version but I haven't been able to.

jnskkmhr, 2024-11-14:
@irfanalimd 
Hey, I haven't be able to test stereo-inertial SLAM since June. 
The current issue is that pose jumps many times probably because of stereo-imu calibration. 
Stereo SLAM runs, but the scale of pose seems not correct (relatively big gap from T265 odometry). 
My repository is here: https://github.com/jnskkmhr/orbslam3
It comes with docker, so that I do not need with build issue.

SrOdranoel, 2024-11-30:
Hello @irfanalimd  and @jnskkmhr.
I have been trying for the past few weeks to run the stereo-inertial system using two Point Grey cameras from FLIR and the STIM 300 IMU. I have already used Kalibr to calibrate my system, which generates a rotation matrix that makes sense. I also used CAD to obtain the equipment positions, and the result from Kalibr makes sense. However, I am still encountering the same problem as mentioned by @jnskkmhr, as detailed below:
```
big time difference
big time difference
big time difference
big time difference
big time difference
big time difference
big time difference
big time difference
big time difference
big time difference
big time difference
big time difference
big time difference
[ros2run]: Segmentation fault
```
I believe the issue is likely related to some change that might have occurred in the repository in recent months. I will try to contact @zang09 and attempt to find a solution. If any of you have found a solution, I would appreciate it if you could get in touch with me. If you'd like to collaborate directly and share information, feel free to reach out to me via email at: leonardo.sales@labmetro.ufsc.br.

If I find a solution, I will reply here or in another issue, and if an urgent solution is needed, I can directly contact you if required.

jnskkmhr, 2024-11-30:
@SrOdranoel 
Hi!
Are you now using this repo? https://github.com/jnskkmhr/orbslam3
In my case, this big time difference came from imu subscriber qos setting. 

https://github.com/jnskkmhr/orbslam3/blob/74d31fda8d80c03d9e4b1c97c7392426c3bcbc49/src/stereo-inertial/stereo-inertial-node.cpp#L61-L64

jnskkmhr, 2024-12-05:
@SrOdranoel 
Hi, it is certainly possible to use my repo without docker. 
You just need to modify cmakemodule path here to your own path: 
https://github.com/jnskkmhr/orbslam3/blob/74d31fda8d80c03d9e4b1c97c7392426c3bcbc49/CMakeLists.txt#L11
https://github.com/jnskkmhr/orbslam3/blob/74d31fda8d80c03d9e4b1c97c7392426c3bcbc49/CMakeLists.txt#L5

Also you might need to deal with different OpenCV version if you work on Ubuntu22.04. 

Btw, I have not solved the lost pose problem as you mentioned, I thought this was due to calibration issue. 
As far as I know, "big time difference" is corrected in my repo (this was due to imu subscriber qos setting). 
The problem may persist if your imu and stereo has big time difference.

SrOdranoel, 2024-12-11:
Hi Again @jnskkmhr. 
After reading and installing your repository, I'm having some doubts and difficulties creating the launch and yaml files (not the yaml files with camera parameters, I already got them from Kalibr without any problems).

Could you help me create the files? I didn't understand the issue of the odometry used. Do you use the IMU as an odometry frame? Why not use remap in the ROS topics?

If you don't mind scheduling a conversation via meetup or discord, my time zone is GMT -3.

jnskkmhr, 2024-12-11:
Sure, @SrOdranoel 
My time zone is in EST. 
Let me know what time works for you. 
My email is jkamohara3@gatech.edu 
You can share me your discord username there.

SrOdranoel, 2024-12-16:
Hi @jnskkmhr,
I have sent you an email using the address leonardo.sales@labmetro.ufsc.br. It might have ended up in your spam folder, but I included my Discord and some of my questions in the message. Please check when you have a chance.

Thank you!

jnskkmhr, 2024-12-28:
@SrOdranoel 
I answered your questions via email!
Please let me know if you have further questions.