zang09/ORB_SLAM3_ROS2 #37: Crashing double free or corruption (out)

CharlieRusso001, 2024-08-22:
When using this wrapper I run into the error below when running with a Realsense D405 camera.  I do now know why I am getting the error double free or corruption (out) which I do not know what this means.  Is there any fix to this?  Attached is also my config file.

charl@charl:~/VO/ROS$ ros2 run  orbslam3 rgbd /home/charl/VO/StereoFixed/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/charl/VO/ROS/src/orbslam3_ros2/config/rgb-d/d405-t.yaml

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D
Loading settings from /home/charl/VO/ROS/src/orbslam3_ros2/config/rgb-d/d405-t.yaml
Camera1.k3 optional parameter does not exist...
	-Loaded camera 1
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
	-Loaded image info
	-Loaded RGB-D calibration
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 672.1 672.1 320.5 240.5 ]
	-Camera 1 distortion parameters: [  0 0 0 0 ]
	-Original image size: [ 1280 , 720 ]
	-Current image size: [ 1280 , 720 ]
	-Sequence FPS: 90
	-RGB-D depth map factor: 1000
	-Features per image: 1250
	-ORB scale factor: 1.2
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
Shutdown
Saving keyframe trajectory to KeyFrameTrajectory
.txt ...
double free or corruption (out)
[ros2run]: Aborted
[d405-t (copy).yaml.txt](https://github.com/user-attachments/files/16703083/d405-t.copy.yaml.txt)

CharlieRusso001, 2024-08-22:
Update:
Tested with a monocular camera as well just to see if the Realsense SDK streaming the camera data I had set up was giving an issue and it seems to output the same segmentation error any clue how to fix?
charl@charl-XPS-8950:~/scald/VO/ROS$ ros2 run orbslam3 mono /home/charl/scald/VO/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/charl/scald/VO/ROS/src/orbslam3_ros2/config/custom/mono.yaml

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- Image scale: 1
- fx: 920
- fy: 920
- cx: 960
- cy: 540
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
There are 1 cameras in the atlas
Camera 0 is pinhole
slam changed
============================ 
Starting the Viewer
[ros2run]: Segmentation fault

Kaushek-31, 2025-04-15:
I have the same issue, did you get any solution for this?

liam8846, 2025-07-23:
It seems that this method didn't work. The attached file is my configuration file.

[astra.txt](https://github.com/user-attachments/files/21379990/astra.txt)