zang09/ORB_SLAM3_ROS2 #38: saving maps

nishantomnipotent, 2024-09-04:
how can i save maps and load the setup in localization mode only

HJ7021, 2025-09-04:
> how can i save maps and load the setup in localization mode only

Hello ,I am currently working with ORB-SLAM3 and I also want to save the map and later reload it just for localization.
Did you manage to solve this problem? If so, could you please share how you did it?

liam8846, 2025-09-05:
> > how can i save maps and load the setup in localization mode only
> 
> Hello ,I am currently working with ORB-SLAM3 and I also want to save the map and later reload it just for localization. Did you manage to solve this problem? If so, could you please share how you did it?

Add the following to the camera configuration file:
System.SaveAtlasToFile: "map.osa"
System.LoadAtlasFromFile: "map.osa"