zang09/ORB_SLAM3_ROS2 #40: Realsense2 D435i camera mono mode crashed

racket405, 2024-11-12:
[image]
I used jetson orin NX 8GB and ROS2 foxy to run ORB-SLAM3 in real time, my camera is D435i,
each time I run `ros2 run orbslam3 mono /home/jetson/vslam_ws/ORB_SLAM3/Vocabulary/ORBvoc.txt/home/jetson/vslam_ws/ORB_SLAM3/Examples/Monocular/RealSense_D435i.yaml`，I got the results above. Then the whole programme just crashed, please help me thanks!

AMLIFY, 2024-11-15:
Maybe you should check if your camera's device format and pixel format is in correspondence with the param.yaml

racket405, 2024-12-11:
> Maybe you should check if your camera's device format and pixel format is in correspondence with the param.yaml

I thought the RealSense_D435i.yaml could be directly used without extra adjustments, what parameters should I check,please?